BarbourSmith/Abundance #332: The output of the input type atom does not behave correctly

BarbourSmith, 2025-05-22:
The input atom has an output which does not behave correctly. The input atom is always locked to the left side of the screen, but if dragged the input atom's output will move away from the left side of the screen as shown circled in green here:

[image]

The input atom also can have it's output hidden when the atom is resized like this:

Correct: 

[image]

Incorrect: 

[image]

alzatin, 2025-06-17:
I'm going to close this since I believe this was resolved by PR #433, but if you think it hasn't, feel free to reopen.

BarbourSmith, 2025-06-17:
Awesome!!

BarbourSmith, 2025-06-17:
@alzatin It’s still looking a little funky on mobile:

[image]

PR #333 (2025-05-22): Fix input atom output connector position and visibility issues
## Issue
The input atom's output connector was not behaving correctly in two scenarios:

1. When the input atom was dragged, its output connector would move away from the left side of the screen instead of staying properly attached to the atom
2. When the input atom was resized, its output connector could become hidden

## Solution
Modified the `draw()` method in the Input class to:

1. Explicitly position the output attachment point on the right side of the input atom
2. Use `GlobalVariables.constrainToCanvasBorders()` to ensure i [... cut]
closed 2025-06-20 by BarbourSmith