BarbourSmith/Abundance #364: Issue with rounded rectangle loading rotated.

BarbourSmith, 2025-05-27:
For some reason this one rounded rectangle always loads off by 90 degrees. Changing any of the inputs seems to fix it.

[image]

In this project: [https://abundance.maslowcnc.com/Abundance/run/BarbourSmith/Chair-Back-Fix](https://abundance.maslowcnc.com/run/BarbourSmith/Chair-Back-Fix)

alzatin, 2025-06-16:
@BarbourSmith 
did you mean to use two different github molecules? 

[image]

alzatin, 2025-06-16:
i think only one is reflecting the problem but i wan't sure if they were supposed to be doing different things

alzatin, 2025-06-16:
Seems to be an issue of shared ID's

alzatin, 2025-06-29:
This is being addressed by other issues dealing with code atom input behavior.

closed 2025-06-29 by alzatin