BarbourSmith/Abundance #672: Puppeteer screenshot taken before project loads

alzatin, 2025-07-27:
In some projects the puppeteer screenshot is taken before the project fully loads making the images appear different. This is caused by the molecule-rendered div being created before the output molecule is compute. 

[image]

Puppeteer should wait for projects output to compute before taking a screenshot

BarbourSmith, 2025-07-29:
I’ve seen this as well. I believe that this is caused by bugs in the way that we are propagating information between atoms. @tristan-huber is working on a major refactoring of how how that happens which should hopefully fix a lot of these type of issues or at least make them easier to debug

BarbourSmith, 2025-08-06:
Ah interesting! That makes sense!

closed 2025-08-07 by alzatin