BarbourSmith/Abundance #819: Molecule with no input field

BarbourSmith, 2025-09-04:
[image]

I just saw something strange. I have a molecule with a "Wood Thickness" input, but when I click on it I don't have an option to edit wood thickness. 

I'm not sure what would cause that, but it's something to keep an eye on.

alzatin, 2025-09-10:
@BarbourSmith I think this is an input type issue. If you have an input type set as geometry like you have in this molecule, the IO will appear but it won't show up as an editable input in the parameter menu becuase we exclude all the geometry inputs so that users can't mess with the unique IDs being passed. I thought the default inputType was number so unless you change it, it should be good and appear as editable but i'll double check and make number the default if it's not.

[image]

BarbourSmith, 2025-09-10:
Ohhh that makes sense! I feel like that might confuse some users so we should find a way to make it clear what is going on...what if input types were color coded?

alzatin, 2025-09-10:
Ah yeah that's a great idea !