BarbourSmith/Abundance #891: Dramatic speed decrease

BarbourSmith, 2025-09-16:
One of the changes today introduced a dramatic speed decrease. Projects are loading in the same amount of time, but then making changes to the project later are dramatically slower.

It could be related to this error message that I am seeing:

```
Error in atom: Move [index-BhSnTQbg.js:143:22900](https://abundance.maslowcnc.com/assets/index-BhSnTQbg.js)
Error: Failed to move geometry: can't access property "geometry", r is undefined
    Jw https://abundance.maslowcnc.com/assets/worker-qz8bUCVl.js:38
[index-BhSnTQbg.js:143:2294](https://abundance.maslowcnc.com/assets/index-BhSnTQbg.js)
```

tristan-huber, 2025-09-16:
Huh, yup. That is something that I removed as part of the refactor earlier this summer. I'm happy to work on re-instituting it.

Did this speed decrease definitely change today? If so there may be a more proximate cause as well. If it's just something that we maybe didn't notice in the last couple weeks then it's probably just my diff which caused it.

alzatin, 2025-09-17:
mhm It would probably be useful to implement something that keeps track of how long things are taking so that we know for sure at what point there's a speed decrease

tristan-huber, 2025-09-17:
PR to restore this optimization at: #899

closed 2025-09-18 by BarbourSmith