BarbourSmith/Abundance #940: Rotation of 2D shapes is broken

BarbourSmith, 2025-09-22:
If you take a circle and move it in the X direction and then try to rotate it it will act in a weird way. I think that something is going wrong with it's plane, possibly related to the recent update to how information is passed between atoms.

tristan-huber, 2025-09-22:
Whoops, definitely my bad. Broke this during the refactor of rotation logic.

Fixed in #941

I have a separate open question about expected rotate behavior: ie should it rotate around the origin (current behavior) or should it rotate around object center? I think the former prefers a rotate-then-move workflow. I'm also not totally conceptually clear on how rotations work for drawings that have already been rotated. Eg: rotate a drawing in y then rotate in z and (IICU) it'll rotate around the normal axis of it's rotated plane. I think this is fine but I'm unclear what happens if I then rotate in x. All of these are low priority things though I think.

closed 2025-09-22 by tristan-huber

BarbourSmith, 2025-09-22:
I think that we want to rotate around the origin, but I am working on adding github atoms which do things like "above" and then if you give it two inputs it will move one up so that it's bounding box is above the other one. If there is no second input it should just move the shape up so that it's bounding box is sitting on the XY plane. I think that a "Center" atom would be nice to have which moves a shape to the center.

PR #941 (2025-09-22): Fix rotate action for planar objects
Fixes #940